Orange-OpenSource/oorobot #4: Comment obtenir le schéma de branchement ?

ChristianM39190, 2017-10-26:


matthieusalvat, 2017-10-26:
Bonjour Christian,

Je viens de rajouter une partie "Le montage" sur la page principale. J'ai aussi mis à jour les fichiers STL pour l'impression 3D.

A noter que ta fille pourra trouver un manuel d'explication et des plateaux à imprimer ici : https://github.com/Orange-OpenSource/oorobot/tree/master/doc/exercice1 

Je suis intéressé pour avoir des retours sur l'utilisation du robot. Ma conjointe aussi; Elle est formatrice à l'ESPE mais aussi ERUN (ex EACTICE) ... ta fille comprendra ces sigles  ;-).

ChristianM39190, 2017-10-26:
Merci pour cette réponse. Je vais commander les composants qui me manquent et ainsi je pourrai peut-être le rendre opérationnel début 2018.
Je vais essayer de penser à faire un retour sur l'utilisation en ULIS.
Merci pour tout et pour ce travail.

matthieusalvat, 2018-05-14:
La documentation a été mise à jour. Voir https://github.com/Orange-OpenSource/oorobot/blob/master/2d/montage.pdf

closed 2018-05-14 by matthieusalvat