Orange-OpenSource/oorobot #16: Licence

Gad, 2019-02-26:
Bonjour,
J'ai bien vu que la licence est GPL 2.0 mais je ne trouve pas d'info sur la licence de fabrication (comprendre : découpe laser) : je suis professionnel de la découpe laser et suis intéressé par proposer votre robots (uniquement la partie mdf) sur mon site qui comprend une partie boutique. Ai-je le droit ? La raison : j'ai reçu des demandes spontannées car tout le monde n'a pas un fablab à proximité et certains fablab interdisent la découpe de MDF 3mm (résines toxiques...)?. Vous pouvez me contacter via le site si vous voulez : Damengo
Merci :)

matthieusalvat, 2019-02-28:
Bonjour,

Pas de problème pour vendre des kits de découpe, la licence choisie (CC-BY-SA https://creativecommons.org/licenses/by-sa/2.0/deed.fr) n'est pas restrictive sur ce point. Par contre vous devrez créditer "l’œuvre" (mettre un lien vers ce site) et partager dans les mêmes conditions d'éventuels modifications du plan.

A ce propos, je travaille actuellement sur une nouvelle version du châssis, le but étant :
- De placer les moteurs vers l'extérieur afin d'avoir plus de place pour bien encastrer les roues sur les axes des moteurs
- D'avoir des éléments permettant de guider les élastiques afin de simplifier le montage des pneus
- De pouvoir modifier légèrement la hauteur de l'axe des roues pour compenser des problèmes de découpes (MDF trop épais, découpe trop large, roues "billes" trop grande)
- De changer l'emprise de découpe de 38cm*38cm à 56cm*38cm afin que l'on puisse utiliser les 2 découpeuses du FabLab de Lannion pour faire des kits.

J'ai découper cette version hier au FabLab de Lannion, je vais la monter au plus tard ce week-end, je vous invite donc à attendre un peu avant de proposer au public la découpe.

Sinon, je suis super content que des kits de découpes puissent être disponibles à la vente sur Internet ... Le FabLab de Lannion reçoit beaucoup de demandes venant d'en dehors du département et que l'on ne peut pas traiter (on ne fait pas d'expédition par courrier).

A voir si vous pouvez proposer la découpe en PMMA : https://twitter.com/matthieu_salvat/status/1087792833073958912 (c'est plus cher mais beaucoup plus solide et joli).

Matthieu

bbailleux, 2019-03-01:
Hello,

Selon les termes de la licence, il est nécessaire de :

> « **Attribution** — Vous devez créditer l'Œuvre, intégrer un lien vers la licence et indiquer si des modifications ont été effectuées à l'Œuvre. »

Pour cela, je suggère de joindre à chaque kit un papillon papier comportant ces informations.

Gad, 2019-03-01:
Bonjour,
Merci pour vos réponses. Bien entendu je n'ai aucun soucis à créditer l'oeuvre, ça me semble plus que normal. De plus mettre un lien vers le github est aussi nécessaire d'autant que les instructions et BOM à jour y sont. 
J'ai bien lu votre avertissement concernant la nouvelle version et attendrai de voir. 
Concernant le plexi, j'ai plusieurs couleurs disponibles en 3mm mais toutes opaques et aucune qui ne soit "frosted" comme dans votre exemple sur twitter : incolore (transparent), noir, rouge, bleu, jaune, blanc diffusant (brillants). C'est vrai que c'est plus sympa mais je ne suis pas sûr que ce soit plus solide : le plexi est cassant et mon MDF très dense (plus dense que celui de Leroy merlin).  Je ferai un essai
Concernant les fichiers 2D il y a un détail que je ne comprends pas (les versions svg et dxf ont le même soucis) : il y a des traits verts que je ne m'explique pas. Pouvez-vous me dire ce à quoi ils correspondent ?
Cordialement,
Gilles

matthieusalvat, 2019-03-01:
Les traits verts sont juste là pour aider à la conception du modèle (centrage de cercle, emprises des cartes électroniques ...). Quand j'importe le DXF sur le logiciel de la découpeuse laser, je dis juste de ne pas prendre en compte cette couleur (output=No sur les machines KeyLand du FabLab).

magsout, 2019-04-08:
@Gad bonjour, (désolé de passer par cette issue). Proposez-vous la découpe sur votre site finalement?

Gad, 2019-04-09:
Bonjour,
Oui, je suis en train de faire le montage de la version 21 (demain après midi) et je le proposerai ensuite si tout va bien.

closed 2019-10-08 by matthieusalvat