Orange-OpenSource/oorobot #17: Nanos WAVGAT

mgenet, 2019-09-17:
J'ai malencontreusement acheté des nanos WAVGAT (https://fr.aliexpress.com/item/32868170637.html) qui apparemment ne sont pas 100% compatibles… (Cf. https://forum.arduino.cc/index.php?topic=560692.0)

Si j'essaye de compiler le code, j'obtiens l'erreur suivante:
```
Arduino: 1.8.9 (Mac OS X), Board: "WAVGAT NANO 3.0"

/.../OoRoBot/arduino/oorobot/oorobot.ino: In function 'void saveParams()':
oorobot:807:10: error: 'class EEPROMClass' has no member named 'put'
   EEPROM.put(0, params);
          ^
/.../OoRoBot/arduino/oorobot/oorobot.ino: In function 'void loadParams()':
oorobot:813:10: error: 'class EEPROMClass' has no member named 'get'
   EEPROM.get(0, savedParams);
          ^
Multiple libraries were found for "Servo.h"
 Used: /.../Arduino/libraries/Servo
 Not used: /Applications/Arduino.app/Contents/Java/libraries/Servo
exit status 1
'class EEPROMClass' has no member named 'put'
```

J'ai essayé de mettre un dossier EEPROM "officiel" (https://github.com/arduino/ArduinoCore-avr), j'ai l'impression que ça compile, mais j'ai l'erreur suivante au téléversement:
```
Arduino: 1.8.9 (Mac OS X), Board: "WAVGAT NANO 3.0"

Sketch uses 17664 bytes (59%) of program storage space. Maximum is 29696 bytes.
Global variables use 1370 bytes of dynamic memory.
the selected serial port avrdude: stk500_cmd(): programmer is out of sync
 does not exist or your board is not connected

avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x94
avrdude: stk500_cmd(): programmer is out of sync
```

Quelqu'un a-t-il réussi à utiliser les nanos WAVGAT? Toute aide sera la bienvenue. Merci!

matthieusalvat, 2019-10-08:
Bonjour,

Désolé de te répondre si tard, mais je ne connais pas du tout les Wavgat Nano. Ca ressemble à un problème que j'avais avec mes Arduino Nano que je commande : as-tu essayé de changer le type de processeur ? Genre mettre ATmega328P (Old bootloader) ?

Matthieu

mgenet, 2019-10-08:
Le vendeur a fini par reconnaître que ses nanos n'étaient pas 100% compatibles, et me rembourser. J'en ai commandé d'autres, avec processeur ATmega328P, qui fonctionnent parfaitement. Moralité: éviter les nanos WAVGAT!

closed 2019-10-09 by matthieusalvat

mgenet, 2019-10-10:
Sincèrement je crois avoir tout essayé…