Orange-OpenSource/oorobot #50: Application OoRoBot Android

Jean-Pierre-Aublanc, 2023-03-09:
Bonjour,

Depuis un smartphone Samsung Galaxy A50 version Android v11 l'appli Android OoRoBot fonctionne.
Mais depuis une tablette Galaxy S6 Android v12 ou Galaxy Tab A7 Android v12, après ouverture de l'appli Android, celle ci se referme aussitôt.

Cdt

matthieusalvat, 2023-03-10:
Bonjour,

Pouvez-vous essayer la manipulation suivante sur votre tablette :
- Ouvrir les paramètres de la tablette
- Aller dans le menu "Applications"
- Sélectionner l'application "OoRoBoT"
- Ouvrir l'entrée "Autorisations"
- Vérifier que le paramètre "Appareils à proximité" est bien autorisé
- Si ce n'est pas le cas il faut autoriser ce paramètre "Appareils à proximité" en cliquant dessus

Merci de me tenir au courant si ça marche (ou pas).

Cdlt

Jean-Pierre-Aublanc, 2023-03-10:
Bonjour,

Après vérification, le paramètre "Appareils à proximité" est déjà autorisé.
En fait, à l'ouverture de l'appli, après qq secondes, celle-ci ne se ferme pas mais se mets en tache de fonds. Si on la réaffiche, elle se remet en tache de fonds après qq sec. etc.

Cdt,

JPA

Jean-Pierre-Aublanc, 2023-03-13:
Bonsoir,

J'ai désinstallé l'ancienne appli et installé la version 1.1.
Résultat : l'appli s'ouvre bien avec les différents menus qui s'affichent.
Au préalable, j'ai autorisé les appareils à proximité ds les paramètres.
Par contre dans le menu Bluetooth, quand je sélectionne OoRoBot ds la liste des appareils, de nouveau, l'appli se met en icone.

Cdt,

JPA