Orange-OpenSource/oorobot #54: clavier 4*4

topi-c, 2025-03-10:
Bonjour 
Le clavier 4*4 que vous proposez n'est plus en fabrication aurez des solutions pour remplacer le clavier par un autre type ainsi que sa documentation de programmation pour le oroobot ?

matthieusalvat, 2025-03-11:
Bonjour,

On peut le remplacer par un clavier tactile de ce genre : https://fr.aliexpress.com/item/1005003658190273.html (ou https://fr.aliexpress.com/item/1005003922142647.html) avec l'option "with PH2.0 Dupont" (branchement sur les ports i2c comme l'écran). De mon côté j'ai fait les touches avec de la découpe de vynile mais une impression sur feuille collée avec du scotch devrait suffire. 

Je vais mettre à jour le code (et le plan de découpe et la doc) cette semaine pour prendre en compte ce matériel (je l'ai déjà fait sur une base d'esp32).

topi-c, 2025-03-11:
Bonjour 
Merci pour votre réponse rapide , je voudrai savoir s'il etait toujours d'actualité pour commander les pieces 3d (kit) ?