Orange-OpenSource/oorobot #61: Erreur Code oorobot.zip

Elecnic1, 2025-05-20:
Bonsoir je vous envoie ce message car je reçois un message d erreur lorsque je transfert le fichier actuelle oorobot.zip avec le nouveau clavier I2C

 avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe0
Failed uploading: uploading error: exit status 1

Merci de votre aide et bonne soirée

Elecnic1, 2025-05-26:
Bonsoir 
grâce a vous , j'ai corriger le problème avec succès , j ai téléverser int-bluethooth avec succès et aussi oroboot.zip sur platformio mais lorsque je lance le moniteur série j'ai ce dernier message qui s'affiche : 

Setup I2C found 39 
Analog keypad not found , use touch screen

j'ai bien l écran led avec le message suivant : Oorobot 1.2.0 Prêt à démarrer ! mais quand j appuye sur un bouton du clavier rien se passe 

Merci infiniment pour votre aide et bonne soirée
Cordialement

Elecnic1, 2025-06-02:
Bonjour , pas de soucis 

merci encore une fois

jblb, 2025-06-02:
j'ai du code, basé sur celui d'oorobot de base qui marche avec les claviers V3, il est [là](https://mygit.jblb.net/jerome/oorobot). Ça fonction avec le clavier branché sur les port A4 pour SDA et A5 pour SCL, le code prend aussi en charge un buzzer sur le port 2

matthieusalvat, 2025-06-03:
Merci Jérôme,
Je me baserai sur ton code quand j'aurai reçu mon nouveau clavier.